MarginallyClever/Makelangelo-firmware #1: AMS won't compile

krummrey, 2015-10-24:
Arduino: 1.6.4 (Mac OS X), Platine: "Arduino Uno"

Build-Optionen wurden verändert, alles wird neu gebaut
Verwende die Bibliothek SPI im Ordner: /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/SPI 
Verwende die Bibliothek AFMotorDrawbot im Ordner: /Users/jan/Documents/Arduino/libraries/AFMotorDrawbot (legacy)
Verwende die Bibliothek Wire im Ordner: /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/Wire 
Verwende die Bibliothek Adafruit_MotorShield im Ordner: /Users/jan/Documents/Arduino/libraries/Adafruit_MotorShield (legacy)
Verwende die Bibliothek Servo im Ordner: /Applications/Arduino.app/Contents/Java/libraries/Servo 
Verwende die Bibliothek SD im Ordner: /Applications/Arduino.app/Contents/Java/libraries/SD 
Verwende die Bibliothek EEPROM im Ordner: /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/EEPROM 

/Users/jan/Documents/Arduino/libraries/AFMotorDrawbot/AFMotorDrawbot.cpp: In member function 'void AF_Stepper::onestep(uint8_t)':
/Users/jan/Documents/Arduino/libraries/AFMotorDrawbot/AFMotorDrawbot.cpp:256:10: error: return-statement with a value, in function returning 'void' [-fpermissive]
   return currentstep;
          ^

i-make-robots, 2015-10-24:
"Verwende die Bibliothek SPI im Ordner" ?

virtuoushub, 2015-10-26:
@krummrey it looks like you need to add Adafruit's V1 motor shield library to your Arduino libraries and `#include <AFMotor.h>` somewhere in your `firmware_ams/firmware_ams.ino`. See https://learn.adafruit.com/adafruit-motor-shield/library-install on where to get the motor shield library. 

I also opened a pull request, see #2 to address the issue in the master branch.

@i-make-robots http://translate.google.com/translate?hl=&sl=de&tl=en&u=https%3A%2F%2Fgithub.com%2FMarginallyClever%2FMakelangelo-firmware%2Fissues%2F1

closed 2015-10-26 by i-make-robots

virtuoushub, 2015-11-07:
Why so serious? b08ec86e160118678eb30fca31a8fc6825df5407

krummrey, 2016-01-10:
Trying to compile again no success. What version of Arduino am I supposed to use? Would love to try the software, but I just can't get the firmware to compile.

i-make-robots, 2016-01-10:
What's the error message?

PR #2 (2015-10-26): Added missing include, and documentation on where to get the library.
Closes #1
commit 4e195249 (2015-10-26): Added missing include, and documentation on where to get the library.
Closes #1