MarginallyClever/Makelangelo-firmware #5: Slow drift upward over time

i-make-robots, 2016-05-16:
Still working on a test case. Most apparent in DXF files. Unique feature seems to be lots of long moves with pen up. Over 10000 commands drift on an A2 drawing will be between 5 and 10 mm, consistent. Problem is repeatable. Not related to weight of counterweights, weight of pen, pulley tooth count, or floor() vs lround() in makelangelo-firmware.  Not related to Makelangelo software.  Seems to be some kind of very small incremental error.  A rounding error?  An off by one in step counting?

i-make-robots, 2016-05-16:
A possible fix available for testing in e60195cebeb7c30516ef8d1f1fd39b56456bf151

tjhawkey, 2016-05-24:
Look like the only change in the motorshield/arduino firmware is "SPOOL_DIAMETER" to 1.5 from 4/pi (about 1.27). How does that change the behavior?
Thanks!

i-make-robots, 2016-05-24:
Also MM_PER_SEGMENT is changed in the 3.2.  I've been dealing with drift on a 3.2.  I am testing a 2.5.2 today.

i-make-robots, 2016-06-28:
This problem was probably fixed by 1496511570105ff525bfd225447482e4aefa8ad3.  @tjhawkey can you confirm?

tjhawkey, 2016-06-29:
Thanks. I will test it next week.

i-make-robots, 2016-07-14:
Fixed in e60195cebeb7c30516ef8d1f1fd39b56456bf151 (rumba) and 67236df26df78fd5c0c209943b567d45efdbc61b (ams)

closed 2016-07-14 by i-make-robots