MarginallyClever/Makelangelo-firmware #6: Can't compile

devburke, 2016-05-21:
So for reasons that don't matter, I had to start fresh with installing the firmware, without access to my old copy. I can't seem to get it to compile now.

At first, I'm getting the error that it can't find the Adafruit_MotorShield files. Then when I copy those into the library (out of the firmware_ams folder), I get this error:

`sketch/Adafruit_MotorShield/utility/Adafruit_PWMServoDriver.cpp:18:37: fatal error: Adafruit_PWMServoDriver.h: No such file or directory
 #include <Adafruit_PWMServoDriver.h>
                                     ^
compilation terminated.
exit status 1
Error compiling for board Arduino/Genuino `Uno.``

But running the MotorParty example file in the motor shield library works just fine.

Any advice?

i-make-robots, 2016-05-21:
You should not have to move the libraries anywhere with the latest Arduino.

devburke, 2016-05-21:
If I don’t, I get this error:

/Users/Devin/Documents/Arduino/firmware_ams/firmware_ams.ino:133:34: fatal error: Adafruit_MotorShield.h: No such file or directory
 #include <Adafruit_MotorShield.h>
                                  ^
compilation terminated.
exit status 1
Error compiling for board Arduino/Genuino Uno.

i-make-robots, 2016-05-21:
Try changing #include <Adafruit_PWMServoDriver.h> to #include
"Adafruit_PWMServoDriver.h".  they're in the same directory.

devburke, 2016-05-22:
Got it working. Just had to restart the IDE to load the libraries. D’oh!

I finally got my DIY Makelangelo up and running. I’m moving the gondola around the page now though, and it’s moving very slow. I’ve tried changing acceleration from 20 to 100, changing the speed from 7500 to 20000, and same with the feed rate. All at the same time. Still moves at a snail’s pace. I’m still looking into why this could be.

closed 2016-05-22 by devburke