MarginallyClever/Makelangelo-firmware #7: Nema 17 stepper motor not rotating properly

IDmedia, 2016-05-29:
I've connected two Nema 17 motors from Adafruit to my Arduino and loaded up the firmware. Now I have a working servo, but only one of the stepper motors is rotating properly. The other (connected to M1/M2) only vibrates. I've tested both steppers so I know they work... Do I have to change anything in the firmware or what could cause the motor to not rotate correctly? The cable pairs should be correct as well. I'm powering the circuit using a 12V/2A power supply.

http://i.imgur.com/2RUh0yL.jpg

i-make-robots, 2016-05-29:
If it only vibrates it either doesn't have enough power or one of the wires
pairs is backwards (so one coil in the stepper fights the other coil)

IDmedia, 2016-05-29:
I finally figured it out! Looks like the shield(s) I brought from eBay aren't good at all. There's a short between these two pins: http://i.imgur.com/O28CkQx.jpg

I'll try to get my hands on a second motor shield and give it another try. I'm also a bit worried about the heat it generates... maybe it's a better idea to get the new Adafruit Motor shield v 2.3?

i-make-robots, 2016-05-29:
You might be able to scratch the board to break the short.
We only support the AMS 1.

Will you be contributing software patches?  Then you'd be safer taking the
2.3.

IDmedia, 2016-05-30:
I'll see if I can figure out where exactly the short is.
This is my first real Arduino project, so I guess I'll stick with the AMS 1 for now, but I'll gladly contribute patches if I get the version 2.3 to work!

Closing this issue now. Hopefully others will find this if they buy a cheap AMS 1 off eBay with a short. :)

closed 2016-05-30 by IDmedia

i-make-robots, 2016-05-30:
This is why I only support Marginally Clever made robots.  I test every
part and ship fully assembled so that any problems are not in the hardware.