MarginallyClever/Makelangelo-firmware #8: One stepper disengages

devburke, 2016-06-17:
For troubleshooting reasons, I had to reflash the firmware on the Makelangelo I had working correctly. However, now I notice that whenever I tell the Makelangelo software to move the gondola around, once it does, it keeps the left motor engaged to hold its position, but the left stepper disengages and it drifts.

i-make-robots, 2016-06-17:
Do you have the previous firmware?  I have been unable to reproduce the problem on the machines we're building here.

i-make-robots, 2016-06-17:
I wonder if you have previous firmware so we can diff and find what's changed.

devburke, 2016-06-17:
I just tried it using the latest release. I should've known better than to mess with what's working :)

i-make-robots, 2016-06-17:
@devburke so this issue is fixed for you?  Some people have reported similar and I'd like to know what you did to fix it, please.

devburke, 2016-06-17:
Oh no, sorry, should've been more clear. It's still doing it with the latest release. Right now, I'm not focusing on that, I'm prioritizing figuring out why one of my machines moves really slow. Swapping out parts between the working and broken ones, etc. So far I think it's looking like the shield is the problem. My theory is corrosion from the flux in my solder is messing things up, but we'll see. It could also be a problem with support for that version of the AFMS.

That doesn't say anything with regard to this issue though.

i-make-robots, 2016-06-28:
Possible fix just checked in 1d8b0fadfa7e39ad0435949500c24e5395a1d05c.  Please flash your firmware and try it.  Reason for bug:  My brain fell out of my head.

i-make-robots, 2016-06-28:
Confirmed fixed.

closed 2016-06-28 by i-make-robots