MarginallyClever/Makelangelo-firmware #9: AFMS v2 shouldn't microstep

devburke, 2016-06-17:
The current firmware uses micro steps on the v2 Adafruit Motor Shield. This works, but it is incredibly slow. I'm going to try removing the micro stepping multiplier and changing to single steps.

i-make-robots, 2016-06-17:
Sure.  you'll get faster motion and less accuracy.

devburke, 2016-06-17:
From what I can tell, the V2 shield is incapable of going anywhere near the speeds of the V1 shield with microsteps. I tried to start a drawing, and it said it would take about 5 days. I can set it to single steps, but it’s much louder.

i-make-robots, 2016-06-17:
Try doing a dry run, after a few lines it will quickly begin to correct the
estimate based on actual time elapsed.  If you have any ideas how to get a
better estimate at the start, i'd like to hear it.

V2 shield is bit-banging (one step at a time to get straight lines) and
bottlenecked because of I2C.  It is a lame board we don't officially
support.  The I2C speed can be doubled by fiddling with the internal clock
but it's still very slow.

devburke, 2016-06-17:
I think the estimate was actually accurate. It would take several minutes just to get to the first line. Looks like I'm gonna just have to track down a knockoff of the V1 shield, since Adafruit doesn't sell it anymore.

i-make-robots, 2016-06-17:
I have a few if you want one. *cough cough*

devburke, 2016-06-17:
Ooh cool! How fast are you able to ship? I'm actually headed out on a long trip and hoping to get this done before I leave.

i-make-robots, 2016-06-17:
I can have it in the mail later today or tomorrow morning, depending on
when you place the order.

devburke, 2016-06-17:
I'd love to buy one of yours, but at the speed I need it, shopping looks to get too expensive for me. I found a generic version on Amazon that I can use Prime for, but since hate to take your sale and I've gotten a lot off of your work, I'll make a donation!

devburke, 2016-06-17:
Actually, is there a way to donate?

i-make-robots, 2016-06-17:
Sure.  Get the Makelangelo software or firmware and pay what you want.

devburke, 2016-06-17:
Ah gotcha! One other idea---do the shields you're selling come with pins already soldered on? I could order it and have it shipped to the house of the friend I'm making this for so she could just plug it in. But she won't be able to do any soldering.

i-make-robots, 2016-06-17:
Yes, soldered on.  If you get the UNO I'll even upload the firmware, too.

devburke, 2016-06-17:
That won’t be necessary, got the board already :)

i-make-robots, 2016-06-17:
Great.  I can have it packed in 10 minutes and shipped out tomorrow
morning, with tracking.

devburke, 2016-06-17:
Awesome, thanks!

closed 2017-12-20 by i-make-robots