MarginallyClever/Makelangelo-firmware #12: engage/disengage issues

tjhawkey, 2016-07-24:
I changed the ams motorshield2 firmware to:
Case 17 = motor engage
Case 18 = motor disengage
Program starts with motors disengaged and button says “engage’. (this is correct)
Program works as expected--engaging when you press and disengaging when you press again

While printing, engage/disengage is grayed out.
After print, button says engage, but motors are already engaged.
Need to press twice (engage, then disengage) to disengage motor.
Button then correctly says “engage” and things work as expected afterwards.

If case 17 and 18 are swapped back to original, the buttons are not right when you start the program.

i-make-robots, 2016-07-24:
Thank you for your pull request. Your heart is in the right place.

M17 and M18 are well known gcode commands. Swapping their behaviour breaks everyone else's expectations.

A better fix would be to make sure the software says disengage at end of drawing.  Perhaps also force an engage when the manual (1/10/100) buttons are pressed so that they don't mistakenly appear to be broken.

tjhawkey, 2016-07-24:
You are right--and I was partly confused. The info above is for 17 = engage and 18 = disengage.
Software works as expected when you boot it, but not after it has completed a move (art, converted image, or plain move.
Engage/disengage button might be better grayed out during a move as well since it doesn't do anything during that time
Sometimes I've had the gondola go wacky and wished for a better way to stop it than unplugging the motor shield. If halt worked at all times it might be useful.

Thanks!

i-make-robots, 2016-07-24:
Halt ceases the transmission of new instructions, but does nothing to the
instructions already in the brain of the robot.  In the Makelangelo 3 that
may be as many as 32 instructions queued.

I will mod the manual controls to grey all buttons while a drawing is
running.  Good idea!

closed 2016-07-30 by i-make-robots