MarginallyClever/Makelangelo-firmware #19: Weird stepper timing

misan, 2017-04-15:
Hi Dan,

I have instrumented timer1 interrupt service so I can see what is going on with the different timer values for the ramping up. For a default movement of 10 mm, at 7000 feedrate and 700 default acceleration (not sure about their units) I get the verbose line move below:
 
	S100	*90	A10	T71	D9	

That will require 10 steps for accelerating and 71 steps coasting and 9 decelerating steps if I understand it right. 

Now I have clocked the timer value for each step:

17543.15873.14598.13605.12738.12048.11494.10989.10526.10152. 

285.285.285.285.285.285.285.285.285.285.285.285.285.285.285.285.285.285.285.285.285.285.285.285.285.285.285.285.285.285.285.285.285.285.285.285.285.285.285.285.285.285.285.285.285.285.285.285.285.285.285.285.285.285.285.285.285.285.285.285.285.285.285.285.285.285.285.285.285.285.285.

10152.10526.10928.11428.11976.12578.13333.14184.15267.

So counter value is reduced when accelerating (first line of numbers) and incremented when decelerating (third line) but ...

I do not understand how it goes from 10152 to 285 counts in the transition from accelerating to coast and from coasting to decelerating. My guess is that there is something wrong with the calculation of elapsed time. I replicated the same results using two different platforms so I do not think the glitch is due to my bad idea of using Serial.print from within the interrupt service routine.

I reckon the above quick may cause missing steps every now and then.

i-make-robots, 2017-04-16:
I can't look into it this deeply for at least 3-4 days.  Interesting!
I would think the accel and decel steps would be the same.
I have not seen any missed steps in a long time, it really messes up
drawing in obvious ways.

misan, 2017-04-18:
Still trying to figure it out. I see in your segment_update_trapezoid code that you calculate the number of acceleration steps as:
` int steps_to_accel =  ceil( ( s->feed_rate_max - start_speed ) / acceleration );`

if feedrate units is steps/second and acceleration is steps/second^2 then what you get is the number of seconds but not the number of steps. By using this number as the total accelerating steps you are missing a big portion of acceleration time, so your trapezoid looks a bit more like a hat.

[image]

i-make-robots, 2017-04-18:
I think I've seen that behaviour on the Stewart platform.

Most of this code is a version of Marlin, the firmware for many 3d printers. Maybe the answer is there.

i-make-robots, 2017-12-20:
Is this fixed now in 7.15?

i-make-robots, 2018-10-10:
Assuming this is fixed as I have not heard back in a year.

closed 2018-10-10 by i-make-robots