MarginallyClever/Makelangelo-firmware #21: ZARPLOTTER code missing in RUMBA version

kukomiguel, 2017-08-06:
In the configuration file, there ia definition for ZARPLOTTER, but the code for RUMBA/RAMPS is missing, not using 4 steppers and Inverse Kinematics are calculated for x, y and l1 and l2. z and l3 and l4 are missing.

i-make-robots, 2017-08-06:
it is possible all I did is add a header, but never got around to adding
the IK.  Let me see what I can do.

i-make-robots, 2017-08-06:
> Ph: +1 (604) 259-9564 <(604)%20259-9564>
> Mo: +1 (604) 916-2281 <(604)%20916-2281>

So the big thing is that the code supports 2-motor machines.  Changing all
the code to support 2+ motor machines is a lot more work than I want.
Changes would have to be made to the IK, findHome, recordHome, the ISR, the
planner, and probably other sections.

kukomiguel, 2017-08-06:
Dan, ZARPLOTTER would be very similar to the Skycam with a single arduino that I told you about.

Is it a lot of work? Can I pay for the development? I'm really unable to do it by myself.. don't have the skills..

Thanks Dan

Miguel.

> El 6 ago 2017, a las 18:35, Dan Royer <notifications@github.com> escribió:
> 
> >

i-make-robots, 2017-08-06:
You certainly can pay me!  $200 cad and I can make it happen by the end of the week.

kukomiguel, 2017-08-06:
Thank you Dan, 

I need a Skycam like robot (movement x,y and z), but with three additions: 
1. The possibility of defining the size of of the object to be moved. (see picture bellow)
2. The possibility of changing the location of the of the motors.  (see picture bellow)
3. The possibility of controlling it with a Joystick or buttons (north, south, east, west and up and down).

If you confirm it can be done I send you the payment. 

Thanks a lot.

i-make-robots, 2017-08-07:
I see no picture.  I'm going to add support for four motors and four limit switches on a rectangular table.  what happens after that is a separate issue.

kukomiguel, 2017-08-07:
Ok. Thanks Dan.

Miguel.

i-make-robots, 2017-08-07:
Can you show me a pic of your machine, please? :)

i-make-robots, 2017-08-07:
Implementation attempted in c0c1f823988cdadc91408de50dad619d8b80a409 on the dev branch.  Give it a try and let me know how it works.  If you share your physical design with me I'll try to replicate it here for testing.

i-make-robots, 2017-08-07:
Odds are the zar plotter implmentation can have MUCH high top speed.

kukomiguel, 2017-08-07:
What i'm trying to do is a SpiderCam as you did in 2012 but with a single arduino and be able to control the movements with a joystick. Here you have a picture:

[image]

i-make-robots, 2017-08-07:
Oh.  That's not a ZARPLOTTER at all.  This is: https://www.youtube.com/watch?v=lNC9tqnVhwg

i-make-robots, 2017-08-07:
You probably wanted https://github.com/MarginallyClever/Skycam

i-make-robots, 2017-12-20:
given that some zarplotter code is in the firmware and this is an image of a skycam, I am closing this issue.

closed 2017-12-20 by i-make-robots

mgpc2005, 2018-12-13:
Hi. How can i get more information about the skycam. I uploaded it to Arduino Mega and ramps 1.4. Pls advise.

thanks