MarginallyClever/Makelangelo-firmware #22: Movement with RAMPS is not correct

kukomiguel, 2017-08-24:
Hello Dan, 
I tested dev branch with  RAMPS by changing the board definition:
#ifdef ZARPLOTTER
#define MAKELANGELO_HARDWARE_VERSION 6
#define MOTHERBOARD BOARD_RAMPS

And when I send the command "G0 X20 Y00 F100" NW an SW motors should release cable and NE and SE motors should pull cable. What I see is that Motor NW doesn't move. 

When I send G0 X00 Y20 F100 instead I see SE motor not moving.  This is really strange.

I don't really know why this is happening as with RUMBA board was working correctly as I saw in the video.

i-make-robots, 2017-08-24:
Try individual motor movements.

D0 [Rnn] [Lnn] [Unn] [Vnn]

R is NW, L is NE, U is SW, V is SE.  if one of these motors does not move, then either it's a software problem (wrong pin setting in RAMPS) or a hardware problem (broken wire, dead driver)

Yura80, 2017-08-24:
Ylk0o

i-make-robots, 2017-08-24:
Ugh, I made a typo.  The order is L=NW, R=NE, U=SW, V=SE.

kukomiguel, 2017-08-27:
Hello Dan  I Tried individual motor movements and they move correctly but 
L motor is NE connected to Y in RAMPS (not NW), R motor is NW connected to X in RAMPS (not NE), U motor is SW connected to Z on RAMPS and V motor is SE Connected to E0 in RAMPS

D0 L100 NE motor moves ok
D0 R100 NW motor moves ok
D0 U100 SW motor moves ok
D0 V100 SE motor moves ok

When I send the G0 X10 code NW (L motor) does not move and the other 3 do move ok
When I send the G0 Y10 code  SE (V motor) does not move and the other 3 do move ok.

I setup Feed rate = 100
I don't have a servo connected to the board as I'm just using the form motors to move the camera support.

i-make-robots, 2017-09-01:
I rechecked the pin assignments.  pins for U (E0) are correct.  Are you sure your hardware is good?  Maybe your driver or your RAMPs board has an issue.

kukomiguel, 2017-09-14:
In last version of the dev branch, the code is able to move the 4 motors correctly and coordinated. The only issue I have is that when sending "G0 Xnn" command the plotter moves to the bottom (+X should move to the right) and when I send the G0 Ynn command the plotter moves to the left (+Y should move up)

I thought that:
A) I could have reel in the cable in the wrong direction in some of the stepers, but that cant be the issue because they are moving coordianted, wrond direction, but correct movement. 
B) Axes X and Y could be inverted, but again this is not the problem as

i-make-robots, 2017-09-14:
The very latest dev branch is a major new direction. You can see it is merging in support for deltas and other kinematic systems.  One of the changes is that software inverting of motors is no longer a thing. If a motor runs backwards you'll have to physically reverse the wiring.  A one time nuisance that avoids a lot of code.

i-make-robots, 2017-12-20:
Is this issue now corrected with ramps in 7.15?

closed 2018-06-27 by i-make-robots