MarginallyClever/Makelangelo-firmware #23: End stops + Ramps - deactivates "Find home"

boehmphilipp, 2017-09-12:
Hi, 
if I include endstops in the firmware for ramps boards and attach endstops to X-Min and Y-Min it deactivates the find home option in the Makelangelo Software. If I connect them after the software is started I can use the endstops. Any thoughts?

Best,
Philipp

i-make-robots, 2017-09-12:
sound like they're wired wrong.  I believe the switches are normally open
switches, and you may have them set up as normally closed.  trying to home
would immediately trigger as it thinks the switches are being touched
already.

boehmphilipp, 2017-09-12:
Ok, I'll give that a second look, but it seems odd that they would work ok if I connect the switches after I start the Makelangelo software.

i-make-robots, 2017-09-12:
re-reading your description... yeah, that does sound wierd.  after you
connect w/ software can you still make it move, or has it silently rebooted?

closed 2018-06-27 by i-make-robots