MarginallyClever/Makelangelo-firmware #25: Add menu to draw common paper boundaries

i-make-robots, 2017-12-20:
assumes starting at home position.
A0...A4?
portrait and landscape?

halfluck, 2018-10-03:
+1 this would be fantastic!

i-make-robots, 2018-10-03:
@halfluck I think instead of in the calibration menu it'll be a generator and it will work with whatever paper size is currently set.  That way it will draw the correct border no matter the paper size.

i-make-robots, 2018-10-03:
Ugh, I thought this was in Makelangelo-software.  coffee wasn't kicking in yet.

halfluck, 2018-10-04:
Yes coffee is a very important part of the process :) ==> [https://imgur.com/23kgyYg](url)
I use the Boundary on Makeangelo Software, First I set the page size to A3 etc then do a boundry
so I know exactly where to stick the paper.

i-make-robots, 2018-10-04:
DUDE.  That was great.  tag me in more of these.  I want to share sweet sweet drawings people do.

i-make-robots, 2018-10-10:
committed an implementation in dbee104

closed 2018-10-10 by i-make-robots

halfluck, 2018-10-11:
Hi Dan,

I can confirm this border feature is working!

Cheers Rob.