MarginallyClever/Makelangelo-firmware #34: Mega 2560 + RAMPS + corexy

jsecondo, 2018-03-17:
Hi there,

I'm having trouble configuring and setting this machine. 
It's a COREXY (380mm x 380mm) controlled with a mega 2560 + RAMPS 1.4. 

Edited the configure.h:
#define MACHINE_STYLE COREXY  
and
#define MOTHERBOARD BOARD_RAMPS

When I try to connect the controller 7.15.5, after I hit connect, the program just hangsup. I loose control over it and I can't see anything in the log window
:
When I try to connect with older version like 7.11.0, it seems connected, the log shows this:
HELLO WORLD! I AM C,G04,G28,G90,G91,G92,M18,M114
HELLO WORLD! I AM COREXY #0
Firmware v8
== http://www.marginallyclever.com/ ==
M100 - display this message
M101 [Tx.xx] [Bx.xx] [Rx.xx] [Lx.xx]
- display/update board dimensions.
As well as the following G-codes (http://en.wikipedia.org/wiki/G-code):
G00,G01,G02,G03,G04,G28,G90,G91,G92,M18,M114

After that, stays connected but have no control over the machine. I can load pics and process them but nothing moves. No motor or servo movement. The machine does a low hum and motors are stuck. The ramps board is hot. 

Same machine with grbl moves allright

Any ideas, I've been trying to make your software working for a long time. I just read tha there was support for RAMPS and corexy configuration so I give it another try.

Latests versions on Arduino IDE.

Did some editing to the language file and created spanish.xml so you could use it (instead of piggy latin)

Any ideas, 

Help needed.

https://lh4.googleusercontent.com/go-0s7Hx2HUzCiguHawKsGVjVbGTURoeX4plWP0Ovx73wkNRmkH9WPX_5pSbLLlrc6KUsNlx_PV3L6NrcY4V=w675-h607-rw
https://lh6.googleusercontent.com/zF_A1qk83nY1s7tf7eesem6TCaXIHKLce6J_uXadDRgNRJu0QlOwCVqkSDFf3Tn8EPFFBzyRbPTWmb6w2W38=w675-h607-rw
[spanish.xml.zip](https://github.com/MarginallyClever/Makelangelo-firmware/files/1822023/spanish.xml.zip)

i-make-robots, 2018-03-17:
i believe the software sees "I AM A COREXY" and doesn't recognize the machine type.  Currently IIRC the software does a shitty job of recognizing each machine type.  See https://github.com/MarginallyClever/Makelangelo-software/blob/master/src/main/java/com/marginallyclever/makelangeloRobot/MakelangeloRobot.java#L163  It needs a better way of adding more types and supporting them dynamically?

i-make-robots, 2018-03-17:
PS your links are 403.

jsecondo, 2018-03-18:
Sorry bout pics. I'll upload them to dropbox

I'm a programmer but not a good one in Java. I found the part of code you mentioned but don't know how to recompile the whole project. 

Did another try configuring as POLARGRAPH. (I have both a COREXY and another POLARGRAPH machine). The 7.15.5 software does the same thing, it just hangsup. I loose control after connecting. No text on terminal, no nothing. Motors hum but nothing else happens. If I don't connect the machine, the soft allows all the funcitionality.

I tried again with older versions and the V8 message keeps popping but the older versions of the soft don't hangup. 

Thanks

jsecondo, 2018-03-18:
Shared the pics in Drive.
Please try this

https://drive.google.com/drive/folders/1AnLhoMMwqm3TF5GcSl2_UmUDjSnuh_lV?usp=sharing

i-make-robots, 2018-03-18:
The robot you showed is an AxiDraw.  Does that count as a corexy?  Not relevant.

I don't know why your machine is "loosing" control.  After you connect can you send gcode?

jsecondo, 2018-03-18:
Its an axidraw. Using it with grbl and had to configure it as corexy. Works
fine. Still have to config the servo.

When i tried 7.15.5 as polargraph..after connection i cant do anything with
the soft. No buttons respond. Cant even diaconnect. Im using java 1.8.162.
If i dont connect soft works fine. I can load pics and process them. Gcode
is generated with no problem.

i-make-robots, 2018-03-18:
what does the log say?  it gets hello world, keeps getting ">", but move
buttons and "start drawing" are still disabled?

i-make-robots, 2018-03-18:
https://github.com/MarginallyClever/Makelangelo-software/issues/231

Please update your dev copy of Makelangelo-software and try again.

closed 2018-03-18 by i-make-robots