MarginallyClever/Makelangelo-firmware #43: LCD.h

halfluck, 2018-10-04:
Hey Guys,

I noticed with the new build that the the smart controller settings have  moved to LCD.h and are being ignored in board_*.h, This is ok if you are using a RUMBA but if you use for example RAMPS then the pin assignments are wrong and need to be manually changed where as on previous build the correct pin assignments were grabbed from the board_ramps.h
*scrached my head for a minute or two why this wasn't working for me after upgrading firmware*

Cheers Rob.

i-make-robots, 2018-10-10:
hey, I checked into dev some cleanup that miiiight break RAMPS setup.  please 2x check I left everything ok?

halfluck, 2018-10-10:
Looks like board_rumba.h is missing (under #ifdef LCD_IS_SMART)

// Encoder rotation values
#define BTN_EN1            31   /* Pin on SMART Adapter */
#define BTN_EN2            33   /* Pin on SMART Adapter */
#define BTN_ENC            35  /* Pin on SMART Adapter */

halfluck, 2018-10-10:
also this is in configure.h and lcd.h, may confuse someone if they only choose one. I still can't get it to compile

// only uncomment one of these options
//#define LCD_IS_128X64  // reprapdiscount Full Graphic Smart LCD Controller
//#define LCD_IS_SMART  // reprapdiscount Smart LCD Controller (including XXL model)

halfluck, 2018-10-10:
Ah i got it to compile finally, the sequence seems to be wrong.
So if I define which lcd panel i have in configure.h before I do the board includes (board_*.h) it works.

Makes sense now cause the board_*.h asks if #ifdef LCD_IS_SMART then set some variables etc. 

I think i explained it well enough?

#define MOTHERBOARD BOARD_RAMPS // change this

// only uncomment one of these options
//#define LCD_IS_128X64  // reprapdiscount Full Graphic Smart LCD Controller
#define LCD_IS_SMART  // reprapdiscount Smart LCD Controller (including XXL model)

#include "board_rumba.h"
#include "board_ramps.h"
#include "board_sanguinolulu.h"
#include "board_teensylu.h"

i-make-robots, 2018-10-10:
i had two commits not pushed.  oops.

i-make-robots, 2018-10-10:
Please confirm RAMPS works with the latest layout so I can close this issue.

halfluck, 2018-10-11:
Hey Dan, the dev now compiles but LCD is all screwy when I use the encoder.

I believe the pin assignments are messed up in board_ramps.h under SD CARD shouldn't it be:

#define SDPOWER            -1
#define SDSS               53
#define SDCARDDETECT       49