MarginallyClever/Makelangelo-firmware #45: Printing from SD Card

halfluck, 2018-10-11:
Hey Guys,

I am having a few issues Plotting from SD Card directly after a fresh power on of the machine.
I may be missing a simple sequence of events but I’ve only found 2 ways to get it to work

1.) use Makeangelo Software and connect to prime initial commands then I can print from SD Card

2.) manually add the starting commands to the gcode file (.ngc)
e.g.
M101 A0 T575 B-575
M101 A1 T400 B-400
M101 A2 T170 B10
G92 X0.0 Y183.0
D6 X0 Y183
G0 F2000 A2500

Cheers Rob.

i-make-robots, 2018-10-12:
While attempting to test this I found a bug in software.  have pushed to dev.  Still no results on this yet.

halfluck, 2018-10-12:
Phew! It's not just me misunderstanding the system.

i-make-robots, 2018-10-25:
M06 was inserting a pause.  Also the LCD display was crazy slow.  I have improved both of these and the dev branch should run better now since commit 1511c86.  Can you please test and confirm?

halfluck, 2018-10-26:
board_ramps.h is still wrong 

under #ifdef LCD_IS_SMART it should be

// SD card settings
#define SDPOWER            -1
#define SDSS               53
#define SDCARDDETECT       49

halfluck, 2018-10-26:
ok so the LCD is stable now, and I can scroll the wheel to adjust the feedrate OK but when I go into the menu its impossible to scroll the > its like the screen never refreshes.

i-make-robots, 2018-10-26:
updated again with your pin settings for ramps.  I spent quite some time
getting scroll wheel to work right.

i-make-robots, 2018-10-26:
still getting the funny screen you showed me in the youtube video?

i-make-robots, 2018-10-28:
> The LCD is working now with the correct pin assignment, I just can't choose
> options on the menu to test SD card printing!
>
> I can video that if it helps explain the issue?

let me guess - the dial turning doesn't appear to move the cursor, but
clicking seems pretty responsive?

halfluck, 2018-10-28:
Bingo! I guess you have seen this before ;)

i-make-robots, 2018-10-28:
Yeah. Something about the encoder reading is really slow and often misreport sad not moving.

i-make-robots, 2019-07-30:
Please check the latest dev branch, I've added a "settings" menu at the end of the main list that shows belt length, machine dimensions, and home position.  this should help with debugging a bit.
Also turning the dial is (i hope) more responsive than before.

closed 2020-02-01 by i-make-robots