MarginallyClever/Makelangelo-firmware #48: Zarplotter image creation should include opening pen change.

i-make-robots, 2018-11-30:
if this is a makelangelo-software issue then set to "won't fix".

i-make-robots, 2018-11-30:
https://www.marginallyclever.com/forums/topic/push-the-scroll-button-on-the-lcd-to-start-a-drawing/

Piten, 2018-12-01:
there is a M226 in the gcode so I think it is missing something in the firmware.

i-make-robots, 2018-12-01:
testing now with config MACHINE_STYLE ZARPLOTTER, LCD_IS_SMART, and MOTHERBOARD BOARD_RUMBA.

i-make-robots, 2018-12-01:
uncomment HAS_LCD in configure.h.  right now your panel is probably all white squares because the board doesn't know the lcd exists, so it ignores 226.

Piten, 2018-12-02:
Hi Dan

There is no HAS_LCD in configure.h

Best 
Petter

i-make-robots, 2018-12-02:
> Yes, The LCD show just white squares, in configure.h under LCD panel supported I have uncomment
> #define LCD_IS_SMART // reprapdiscount Smart LCD Controller

Get latest. It should add HAS_LCD and HAS_SD

Piten, 2018-12-02:
I have now downloaded version 6.17.1 but I can`t find HAS_LCD and HAS_SD
are there any newer versions?

i-make-robots, 2018-12-02:
the dev branch of the firmware should have it in configure.h.

Piten, 2018-12-02:
I am using a Ramps board does the dev support it?

i-make-robots, 2018-12-02:
yes.

Piten, 2018-12-02:
Where do I find this release?

i-make-robots, 2018-12-02:
https://github.com/MarginallyClever/Makelangelo-firmware/tree/dev
on the main github page there's a button that says "branch:master".  click
it and change to "branch: dev".  you can also pull from github the dev
branch, but instructions vary with your method.

closed 2018-12-03 by i-make-robots