MarginallyClever/Makelangelo-firmware #50: How to compile for skycam

i-make-robots, 2018-12-13:
thanks

_Originally posted by @mgpc2005 in https://github.com/MarginallyClever/Makelangelo-firmware/issues/21#issuecomment-446965473_

i-make-robots, 2018-12-13:
configure.h: 

#define MACHINE_STYLE SKYCAM

closed 2018-12-13 by i-make-robots

mgpc2005, 2018-12-13:
thanks for the reply. can you pls advise how to control the object movement . sorry but there is not enough info.   N.B the firmware compiled for skycam on arduino mega.

mgpc2005, 2018-12-14:
I sent a G0 command and motors are working fine (X motor for demo) Do i need to set a 0,0,0,0  initialization, like the center of a 1m x 1m x 1m cube.will the firmaware accept input from other arduino with joystick like logitech 3d extreme. will the weight attached move smoothly inside the cube in all direction and the connected strings remain straight.

thanks

i-make-robots, 2018-12-14:
These are not software problems.  Please post these questions to the
marginally clever forums where you can talk to other skycam users AND my
answers will help the whole community.  Thanks!