MarginallyClever/Makelangelo-firmware #51: trapezoidal acceleration wrong

i-make-robots, 2019-01-31:
Behavior:
Line segment is sent to planner;
planner adds line segment to segment buffer;
planner recalculates acceleration between segments;
ISR walks over the segments, obeying the instructions laid out by the programmer.

Symptom:
Sound of motors rises and falls repeatedly while travelling in a long straight line.  This seems to indicate the instrutions between segments bring the speed down and back up, instead of staying consistently high.  Also the top speed observed is very low.  I suspect the planner never lets the motors get to top speed because the segment blocks drop the top speed over and over again.

i-make-robots, 2019-01-31:
> A long straight line is represented by a single block, so it should coast
> at max speed till reaching the deceleration time(step).

When I command a long straight line on a makelangelo it splits the line
into sub-lines.  the sound I'm hearing is probably between segments, where
it is incorrectly calculating acceleration.

i-make-robots, 2019-02-02:
dev 5e49c22 updated with my work so far.  describe_segments() will do just that, and has proven useful in debugging speeds between segments.  I'm still getting very strange behavior since switching to the correct acceleration/distance estimate.

i-make-robots, 2019-02-06:
testing now, appears to be much better.  and higher top speed!  2h drawing now being done in ~45min.

misan, 2019-02-06:
Good news!

closed 2019-05-23 by i-make-robots