MarginallyClever/Makelangelo-firmware #60: When setting the machine limits, the Polargraph stops responding for a long time.

mishafarms, 2019-11-09:
Dan,

     When I connect to my polargraph and set the limits from the Makelangelo software, My polargraph motors start making some noise (but don't move) and the machine is unresponsive to the serial port.

I think I have traced the issue down to the following. After the new limits (which are bigger in my case. the code in motor.c function motor_line, there a place where you access the last segment and get the step_count. You then subtract that value from the steps. for me it is at line 1055. But the issue is that we just reset the limit, the distance_mm in this case is 0, so we don't think we need to move and some of the numbers are inf. Since in a system without the limit change the steps would be the same and we would not try to do anything further.

I have a hack to fix this,
    if ( distance_mm != 0.0 )
    {
        new_seg.a[i].delta_steps = steps[i] - old_seg.a[i].step_count;
    }
    else
    {
        new_seg.a[i].delta_steps = 0;
    }

But I figure you might have a better idea how to fix this.

If you do please let me know, If not I can get a pull request ready.
Thanks for help.
Michael

i-make-robots, 2019-11-11:
so... on limit change step count should be set to ... home position with the new dimensions.
Does it check that there WAS a last segment?  because maybe there wasn't and last segment is uninitialized variables.

mishafarms, 2019-11-11:
There was a last segment. It was 15. I believe that there were commands run when we first connected,
then the limit change. 
And yes to the on limit change the step count should be changed.  But it would have to be on the last segment if there is one.

i-make-robots, 2019-11-12:
What are the exact steps to reproduce the problem WITHOUT the software?

mishafarms, 2019-11-13:
Dan,
    Well I couldn't at first reproduce this without the software. I would send what I saw where the same series of commands but it would not fail.
So what ultimately happened was I found that the host software was sending a few empty lines. Just
'\n'. So that seemed to do it. 
I think I know what is going on. When the code receives the newline, it looks to see if the first char is  ':' and if not then parse the line (no line was read, but the old stuff is still there). So we processed a command that was left over.

Here is my new fix. (I will submit a pull).
the first line of processCommand should be 
  if ((serialBuffer[0] == '\0') || (serialBuffer[0] == ';')) return;  // blank lines
 do not process an empty line. 
Thanks for asking if I could reproduce without the software. I would not have figure it out without that. I thought I had found the bug in the motor code.

     Michael

closed 2019-11-13 by mishafarms

i-make-robots, 2019-11-13:
Michael,

Have you got a website featuring your work?  I love that you've made pull
requests and I want to show my appreciation by getting some eyes on your
talent.