MarginallyClever/Makelangelo-firmware #73: Strangest issue, X and Y motor same motion

Ohmarinus, 2020-02-22:
Hi, I am trying to get Makelangelo work on a homebrew machine running Ramps 1.4

However, the strangest thing is happening. I got it to work once, after that something blew up in my mega board because I had a faulty servo. So I grabbed a new Ramps and Mega board and reuploaded the firmware.

Got it to work again, but wanted to make a change to some firmware settings. So I reuploaded the firmware, and now the motors will only either do the exact same movements, or mirror each other when I swap the stepper motor cable.

Is there a known error where the X and Y motors somehow make the same movements or get the same signal? I have checked the pins file for the Ramps board and it all seems to be correct.

After running into this problem I decided to start fresh with the firmware re-downloaded from GITHUB but somehow the issue persists after reuploading the fresh firmware where I only changed the board to ramps.

Can is be that this freak issue has somehow anything to do with a corrupt EEPROM? I tried yet another Mega with another Ramps board and it has the same problem. I am 100% surely not wiring anything wrong. I have built 8 3D printers in different setups, cartesian, corexy, delta, made a lasercutter, a pen plotter both with corexy and now I am running into this issue. It's so strange!

Here's a video of what's happening:
https://www.youtube.com/watch?v=UUAYNTyOET4

i-make-robots, 2020-02-22:
Did you set the machine size? If the width is zero then the math gets dumb.

Ohmarinus, 2020-02-22:
Thanks for your swift reply. I don't see any place in the firmware where the machine size can be set, also the LCD displays the correct X and Y coordinates. It gets even stranger when I say that the machine has once worked with the firmware I'm using. And I've never set the machine size in the firmware because searching for it hasn't yielded any results.

Could you point out to me where the machine size has to be set? Both searching for 'size' and 'width' didn't give any result for machine size.

i-make-robots, 2020-02-22:
You’re looking for the x and y limits. D11 is a command we use to set default values for the m5.

Ohmarinus, 2020-02-22:
Alright, the issue is that I can connect to the machine with for example Pronterface but I cannot send anything through the serial connection.

When I use Arduino's serial monitor and type D11 it says:

Adjusting limits.
Saving limits.
Saving calibration.
Saving home.

But I do not get any options. Sorry, I hope I am not being annoying!

I now have hardcoded:
`  float limit_xmin = -275;
  float limit_xmax = 275;
  float limit_ymax = 375;`

Into the firmware and hope this will work.

Is there any place I can find out how to use D11 through the serial monitor?

i-make-robots, 2020-02-22:
D11 only sets one set of dimensions. There is another D command to set unique dimensions. In Makelangelo software you could go to settings, set machine type to custom, adjust the size, and then you’d see the output in the log panel at the bottom.

It’s rarely annoying to help someone with robots, especially when they’re eager and willing to make it work like you.

Ohmarinus, 2020-02-22:
Thanks, I appreciate the help!!

So far I've gotten the machine to do this:
https://www.youtube.com/watch?v=Izg4my7MCFQ

Still a lot to do, especially on the mechanical and configuration side. For example, I do not know what the ideal home position would be. Eventually I will add endstops, but for now I want to first make the machine stable and try to get it dimensionally sound. And add weight to the carriage to stabilize it.

i-make-robots, 2020-02-22:
Try the development branch.  it should run smoother than that.

Ohmarinus, 2020-02-23:
Okay, I will search for the dev branch. The LCD also becomes unresponsible during drawing so I suspect the buffer is completely flooded. I tried playing with the segments but haven't been able to fix it yet. There is a very distinct pattern in the outer ring of the resulting drawing which reminds me of a 3D-printer problem called 'salmon skin'.

[image]

Ohmarinus, 2020-02-25:
I have narrowed down the speed issues, it only happens when drawing round curves such as the ones generated by 'Spiral'. When doing the generated artworks with straight lines it all works perfectly and the fonts/texts also render fine.

I'm down to trying to get dimensional accuracy now because an A4 looks more like an A3 chopped in half lengthwise with a bulge on top and a concave bend on the bottom. However, I have set the dimensions perfectly and the steps/mm and pulley diameters are perfect as well so it must be something else. I'll try to find a solution.

The new firmware didn't make a difference speedwise. However, moving from A4988's to DRV8825's that I had laying in the trashbin (sorry) seemed to have improves the noise and precision by a lot. 1/32 microstepping is definitely quieter and better on the ears. I'm still experimenting with how fast I can have the servo actuate but I'm starting to think that using a good quality solenoid might also do the trick. If you or would have any tips for a drop-in (5 to 12v with signal pin) solenoid replacement I'm all ears! :)

i-make-robots, 2020-02-25:
if your machine dimensions are good and your image is still big-on-top then
your starting point is probably too high on the board.

also see:
https://www.marginallyclever.com/2014/10/how-to-fix-9-common-polargraph-drawing-problems/

for more tech talk, please use the forums.  there are lots of people who
will answer faster than I can.  thanks!

Ohmarinus, 2020-02-27:
Yes, I had moved onto the forums, thanks a lot for your help Dan! :)
The reason why I went here was because my account on the forums was awaiting approval for around 5 days and I was in a holiday week and wanted to work on the project. Your advice helped a lot. If I encounter any firmware errors I will report here. For anything else I'll move to the forums!

i-make-robots, 2020-02-28:
You, sir, are a gentleman and a scholar.

I'm sorry I didn't get back to the forums faster.  Too much to do at once :(