MarginallyClever/Makelangelo-firmware #74: WEMOS D1 R2 Arduino IDE

phantomse, 2020-05-24:
Which version of Arduino IDE and Library to use to compile for WEMOS D1 R2 ???

i-make-robots, 2020-05-24:
Are you saying the latest versions don't work?  Did you `#define MOTHERBOARD BOARD_WEMOS` in configure.h?

phantomse, 2020-05-31:
cant find any reference to ither of these. Tried multiple replacement libraryes ...

/home/Downloads/Makelangelo-firmware/Makelangelo-firmware.ino:14:20: fatal error: SDcard.h: No such file or directory
 #include "SDcard.h"
                    ^

- [ ] /home/Downloads/Makelangelo-firmware/Makelangelo-firmware.ino:15:17: fatal error: LCD.h: No such file or directory

 #include "LCD.h"
                 ^

In file included from /home/book/.arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/sdk/lwip2/include/glue.h:53:0,
                 from /home/.arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/sdk/lwip2/include/arch/cc.h:61,
                 from /home/.arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/sdk/lwip2/include/lwip/arch.h:48,
                 from /home/.arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/sdk/lwip2/include/lwip/debug.h:40,
                 from /home/.arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/sdk/lwip2/include/lwipopts.h:3608,
                 from /home/.arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/sdk/lwip2/include/lwip/opt.h:51,
                 from /home/.arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/sdk/lwip2/include/lwip/init.h:40,
                 from /home/.arduino15/packages/esp8266/hardware/esp8266/2.7.1/cores/esp8266/IPAddress.h:27,
                 from /home/.arduino15/packages/esp8266/hardware/esp8266/2.7.1/libraries/ESP8266WiFi/src/ESP8266WiFi.h:31,
                 from /home/Downloads/Makelangelo-firmware-master/Makelangelo-firmware/Makelangelo-firmware.ino:61:
/home/.arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/sdk/include/user_interface.h:34:2: error: #error LWIP_OPEN_SRC must be defined
 #error LWIP_OPEN_SRC must be defined
  ^

i-make-robots, 2020-05-31:
In the header file for wemos add 

#undef HAS_LCD
#undef HAS_SD

it should then stop trying to include these features.

phantomse, 2020-05-31:
In file included from /home/book/.arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/sdk/lwip2/include/lwip/opt.h:51:0,
                 from /home/book/.arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/sdk/lwip2/include/lwip/init.h:40,
                 from /home/book/.arduino15/packages/esp8266/hardware/esp8266/2.7.1/cores/esp8266/IPAddress.h:27,
                 from /home/book/.arduino15/packages/esp8266/hardware/esp8266/2.7.1/libraries/ESP8266WiFi/src/ESP8266WiFi.h:31,
                 from /home/book/Downloads/Makelangelo-firmware-master/Makelangelo-firmware/Makelangelo-firmware.ino:61:
/home/book/.arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/sdk/lwip2/include/lwipopts.h:1305:2: error: #error TCP_MSS must be defined
 #error TCP_MSS must be defined
  ^
/home/book/.arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/sdk/lwip2/include/lwipopts.h:2391:2: error: #error LWIP_IPV6 must be defined
 #error LWIP_IPV6 must be defined
  ^
/home/book/.arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/sdk/lwip2/include/lwipopts.h:3551:2: error: #error LWIP_FEATURES must be defined
 #error LWIP_FEATURES must be defined
  ^
In file included from /home/book/.arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/sdk/lwip2/include/glue.h:53:0,
                 from /home/book/.arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/sdk/lwip2/include/arch/cc.h:61,
                 from /home/book/.arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/sdk/lwip2/include/lwip/arch.h:48,
                 from /home/book/.arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/sdk/lwip2/include/lwip/debug.h:40,
                 from /home/book/.arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/sdk/lwip2/include/lwipopts.h:3608,
                 from /home/book/.arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/sdk/lwip2/include/lwip/opt.h:51,
                 from /home/book/.arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/sdk/lwip2/include/lwip/init.h:40,
                 from /home/book/.arduino15/packages/esp8266/hardware/esp8266/2.7.1/cores/esp8266/IPAddress.h:27,
                 from /home/book/.arduino15/packages/esp8266/hardware/esp8266/2.7.1/libraries/ESP8266WiFi/src/ESP8266WiFi.h:31,
                 from /home/book/Downloads/Makelangelo-firmware-master/Makelangelo-firmware/Makelangelo-firmware.ino:61:
/home/book/.arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/sdk/include/user_interface.h:34:2: error: #error LWIP_OPEN_SRC must be defined
 #error LWIP_OPEN_SRC must be defined
  ^
exit status 1
Error compiling for board WeMos D1 R1.

i-make-robots, 2020-05-31:
These errors are coming from the underlying hardware libraries.  They're
not caused directly by Makelangelo Firmware code.