MarginallyClever/Makelangelo-firmware #75: SANGUINOLULU COMPILE ERROR

ekaggrat, 2020-06-22:
hi,

I am getting the following error when try to compile for sanguinolulu. I intend to use the firmware on a custom atmega 644p based board . 

i get the following error

sketch\motor.cpp.o: In function `__vector_13':

sketch/motor.cpp:1055: multiple definition of `__vector_13'

sketch\MServo.cpp.o:sketch/MServo.cpp:104: first defined here

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board ATmega644.

thanks again for the firmware

i-make-robots, 2020-06-22:
It has something to do with multiple ISR methods. MServo.h is probably including Timerone.h and that conflicts with the timer used to move the stepper motors.

> On Jun 21, 2020, at 7:52 PM, ekaggrat <notifications@github.com> wrote:
> 
> multiple definition of `__vector_13'

ekaggrat, 2020-06-22:
i changed it to timer2 from timer1 in the mservo.cpp file. it now compiles but says too big for 644p . So i tried compiling it for 1284p and it does it. anyway to shrink the sketch. I am trying to fit it on a compact board and hence dont want to use a ramps or any other mcu.

i-make-robots, 2020-06-24:
Hi!

Can you please make a pull request to add your sanguinolulu changes back
into the firmware?  I have other people making Ender3 > makelangelo mods
and they're hitting the same bug.

Thank you!

i-make-robots, 2020-08-05:
closed because inactive.

closed 2020-08-05 by i-make-robots

dreiduke, 2021-03-17:
get the same error