MarginallyClever/Makelangelo-firmware #79: Need github Action for Continuous Integration

i-make-robots, 2020-06-26:
Should compile firmware once for each board time * once for each machine type
That way all pushes and pull requests can confirm they do not obviously break other machines.

i-make-robots, 2020-07-14:
https://github.com/ArminJo/arduino-test-compile seems like the right direction.  I can't grok what's going on to make it work.

per1234, 2020-11-13:
The `ArminJo/arduino-test-compile ` action is an excellent choice, but in case you would like to consider an alternative, I'll also mention Arduino's official action:
https://github.com/arduino/compile-sketches

If you have any questions about how to use it, I'd be happy to answer them.

If you do give it a try, I welcome any feedback, feature requests, bug reports, etc.

i-make-robots, 2020-11-13:
Looks great! I only want a report on fail. I see no way to define preprocessor commands, which was a big sticking point in my current setup.

per1234, 2020-11-13:
You're right. I don't see any way to get the `--build-property` option with `arduino/compile-sketches`, and there is certainly no good way.

I have had it on my "to do" list for a while to add an input to the action to allow the user to specify arbitrary additional flags to use with `arduino-cli compile` but I've been mostly taking the approach of only adding complexity to the API as there is evidence of specific need for it from the users. This is the first report I've seen of someone needing that functionality, so this is some motivation in that direction.

I see that `ArminJo/arduino-test-compile` already has the specific capability of setting the `extra_flags` properties specifically:
https://github.com/ArminJo/arduino-test-compile#build-properties

closed 2020-11-23 by i-make-robots

per1234, 2021-02-08:
I know that you're already set with a nice CI system, but I'll follow up on this conversation by reporting that the `arduino/compile-sketches` action now has the ability to set arbitrary `arduino-cli compile` flags like `--build-property`:
https://github.com/arduino/compile-sketches/pull/18