MarginallyClever/Makelangelo-firmware #83: ESP32 implementation - need help with timers

captFuture, 2020-08-04:
Hi Dan,
After following your project for a long time, I decided to give it a try today and my current setup is based on an esp32 (wemos lolin32) so i forked your repo, moved the files from Arduino IDE to Platformio and started adding the esp32.

Unfortunately now I am stuck with the timer functions and can't get them to work.
These are my last errors before it should at least compile ;)

Maybe somebody can help me.
Thanks in advance
Chris

i-make-robots, 2020-08-04:
I have not done any work on ESP32 so I'm no help there.  try this: https://techtutorialsx.com/2017/10/07/esp32-arduino-timer-interrupts/

captFuture, 2020-08-05:
Thanks for you comment, I'll try to get this right somehow 👍

captFuture, 2020-11-11:
Current dev commit of esp32 works, as far as I can tell (in this branch Mega2680 compiling is not possible, but this is a different issue)

closed 2020-11-11 by captFuture