MarginallyClever/Makelangelo-firmware #84: WEMOS D1 R2 - watchdog reset

captFuture, 2020-08-05:
The latest version (master) goes into a continuous reset loop (resets every 5seconds)

```
ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld
WIFI OK
UDP OK
Use this URL to connect: http://192.168.4.1:9999/
```

Which means the Hardware Watchdog resets (rst cause:4)
I will try an older version to maybe track down the issue - currently i have no clue at all.

i-make-robots, 2020-08-05:
A recent contributor, Miguel S, knows more about the WEMOS stuff than I do.  I've reached out to him and I hope he'll join the conversation.

i-make-robots, 2020-08-05:
@phantomse might also know a thing or two.

i-make-robots, 2020-08-05:
right, Miguel's handle is @misan