MarginallyClever/Makelangelo-firmware #88: Fix for compiling error on D50 D51

captFuture, 2020-11-09:
in parser.cpp and parser.h new commands are introduced but not implemented fully yet.
they lead to compiling errors.

Added
void Parser::D50() {
  // something has to go on here
}

void Parser::D51() {
  // something has to go on here
}

to my capBranch locally after line 582 in parser.cpp

and corrected

  void D50(); 
  void D51();

in parser.h 

There is a missing semikolon at some end of line when compiling as well, but I can not remember now.

During my latest test on the dev branch some hours ago before you merged to master the encoder did not work, i will investigate this. 
(probably will switch to Platformio and use your repo as upstream because library management and debugging is horrible in Arduino Ide, but need to figure out how git-subtree works)

i-make-robots, 2020-11-09:
I know that D50 is meant for "strict" mode - once it is on you MUST send checksum or command will be rejected.
D51... I don't see any notes about what it is for.

captFuture, 2020-11-09:
I do not see anything for D51 either - it just appeared;

BTW: the missing semicolon is in parser.cpp line 817

void Parser::M112() {
  // clear segment buffer
  last_segment = current_segment;
#ifdef HAS_LCD
  LCD_setStatusMessage("ESTOP - PLEASE RESET");
#endif
  // stop clock
  CRITICAL_SECTION_START() --------->missing ; <----------------
  // do nothing
}

i-make-robots, 2020-11-09:
@captFuture  that might be better.

closed 2020-11-11 by captFuture