MarginallyClever/Makelangelo-firmware #89: Encoder movement completely gone

captFuture, 2020-11-11:
In the latest master https://github.com/MarginallyClever/Makelangelo-firmware/commit/328779cd2618d03e5e6818e53480fa988f594073 encoder knob on Rumba board with LCD_IS_SMART active is not working any more.

After trying more debug output in lcd.cpp Line 338 (// for debugging potentiometer) it is visible that the encoder is behaving strange. The movement is detected and right after sending a + value it is instantly sending a - value and the other way around.
- Something might be firing twice

Screenshot: [http://tarantl.com/serial-makelangelo.PNG](http://tarantl.com/serial-makelangelo.PNG)

**Platform (please complete the following information):**
 - Device: RUMBA

I was trying to change the encoder reading to different versions found and changed it to Paul Stoffregens Encoder Library as well for testing, but the error persists. I will investigate further in my fork.

Additional thought because I saw the open Pullrequest -> there is no interrupt support on pins 11 and 12 of Mega2860 as far as i know. (at least on Rumba pin setting it is 11 and 12)

i-make-robots, 2020-11-12:
Uh... I am running the latest here, dial seems fine.  I don't know what to say.

captFuture, 2020-11-20:
Strange, I just tried it with the 128x60 Display and the same happens. I hope my second Rumba board gets delivered soon, then I test with this hardware and give feedback. I do not know if there are other people out there having this issues as well.
When I am trying the encoder standalone with a basic sketch it works.

i-make-robots, 2020-11-22:
I can confirm the encoder is unresponsive in the latest dev branch.

i-make-robots, 2020-11-22:
lcd_turn seems to be wonky.  when I enable verbose output and turn the dial for every 1 I see a matching -1.  sum of all dial turning never seems to leave zero.

i-make-robots, 2020-11-22:
it looks like the encoder is being read wrong?  so every second tick is read as a turn backwards.

i-make-robots, 2020-11-22:
I don't see anything that's changed in the recent history of the lcd.cpp file that would cause BTN_EN2 to be read as high all the time., and yet that's what's happening....

i-make-robots, 2020-11-22:
I'm working my way through old commits to find the one where the encoder breaks.  it's between august 7 (working) and september 11 (broken)

closed 2020-11-22 by i-make-robots

i-make-robots, 2020-11-22:
encoder works, but now homing doesn't.  wtf have i been smoking...?

captFuture, 2020-11-23:
Waking up at 5 in the morning, reading this made me lough :)
We all sometimes doubt our mental sanity ;)

commit 09e386cb (2020-11-22): encapsulated sixi gripper in HAS_GRIPPER define

fixes #89
commit f55230ca (2020-11-22): more fix #89