MarginallyClever/Makelangelo-firmware #92: rename eeprom.h/.cpp to eeprom_wrapper.h/.cpp

cad435, 2020-11-13:
src/eeprom.h/.cpp defines a class named Eeprom, whilst there is the class EEPROM from Arduino HAL. 

This works somehow with Arduino software but is directly agains C/CPP code guidelines as inclusions are usually case insensitive. This results in massive compilation error when compiling with a standard gcc-compiler (in my case invoked via platformIO).

I'd suggest renaming the own header/src file to "eeprom_wrapper" to differenciate between classes as well as the associated class.

- cad435

i-make-robots, 2020-11-13:
That sound great!  Send in a PR.

cad435, 2020-11-13:
#93

cad435, 2020-11-13:
Also, I can only suggest developing that project in PlatformIO...

This firmware is HUGE already, Arduino gets really messy at this point...

i-make-robots, 2020-11-13:
Yes... arduino mega is about the only version that still runs the firmware.  We're looking at getting into platformio, which might be the moment we break away from arduino.  already have an SKR 1.4 on order and will probably have to make the switch then...

closed 2020-11-13 by i-make-robots

cad435, 2020-11-13:
Ah yeah, Arduino did it again, the pull request you merged has some errors, because the arduino compiler cached to much files and gave me no errors... I corrected them all and will make a new pull request. 

Now it should all be working again (Restarting of the arduino IDE clears the cached files)