MarginallyClever/Makelangelo-firmware #95: Switch to PlatformIO

cad435, 2020-11-13:
As this Firmware is very huge Arduino is really messy.

A switch to PlatformIO will give a full-blown, yet easy to use IDE wich - used on a project this size - comes in really in favor. 
This will also bring in a standard gcc-g++ compiler toolchain wich ommits the "caching" problems sometimes happening with arduino.

A working PlatformIO version can be found here: https://github.com/cad435/Makelangelo-firmware/tree/PlatformIO

- cad435

captFuture, 2020-11-15:
Nice :) I did a different approach locally and used makelangelo firmware as a git- subtree

i-make-robots, 2020-11-21:
@cad435 @captFuture has submitted a PR that makes this happen.  Can you tell if it works?  I'm not yet setup to check.

captFuture, 2020-11-21:
I will also add more configurations in the next days locally if i have enough time, to be able to run a complete all board test myself on my system (win10) compiling with library fetching works for rumba and ramps and it compiles in arduino 1.8.13 (all machine style polargraph) and I am planning to check it on my linux machine as well because the makelangelo-software is not really working good on windows (currently I get bad checksum errors upon connecting to the latest firmware...)
I am investigating changing other stuff in platformio.ini as well, like including header files on demand, but there is a bug on windows systems, where it is not reading the filepath correctly.
My solution is not a complete move to platformio for the project, but gives the user the choice (complete backward compatibility to arduino ide). I can write a tutorial how to initally setup the pio project if you like.

cad435, 2020-11-21:
@captFuture in your Branch RumbDev I can't seem to find any platformIO.ini?

captFuture, 2020-11-21:
> @captFuture in your branch RumbDev I can't seem to find any platformIO.ini?

:D as said in the pull request: create new pio project, clone makelangelo-firmware repo into project folder, delete /src folder of created project. use this as content of platformio.ini 

[platformio]
src_dir = Makelangelo-firmware

[env]
monitor_speed = 57600

[env:rumba]
platform = atmelavr
board = megaatmega2560
framework = arduino
build_flags =
-DHEADER_RUMBA
-DPIO
lib_deps = SdFat, TMC2130Stepper, TMCStepper, U8g2, U8Glib, bitbucket-fmalpartida/LiquidCrystal @ 1.5.0

[env:ramps]
platform = atmelavr
board = megaatmega2560
framework = arduino
build_flags =
-DHEADER_RAMPS
-DPIO
lib_deps = SdFat, TMC2130Stepper, TMCStepper, U8g2, U8Glib, bitbucket-fmalpartida/LiquidCrystal @ 1.5.0

... i just did this for backward compatibility with arduino ide
maybe in future everything else could be included in a completely changed project, but this is up to dan, he is the master of all this :D

cad435, 2020-11-21:
Ah, because you use a submodule, I didn't got that your the same person, sry...

cad435, 2020-11-21:
I can confirm that his changes and his pio.ini compiles without any error.

I updated my PIO branch (https://github.com/cad435/Makelangelo-firmware/tree/PlatformIO) to have an "out of the box" branch wich can directly be opened with PlatformIO, you can try it out. 

I don't know if using it in Arduino IDE will break the IDE, I only did a compilation with PIO

But I'd say its a good step in moving into PIO...

cad435, 2020-11-21:
[image]

captFuture, 2020-11-21:
thanks I will check the tree :D 

Sometimes Direct messages would come in sooooo handy on Github ;) https://discord.gg/bsFjqgNU

cad435, 2020-11-21:
@i-make-robots I Agree with @captFuture but would suggest a MarginallyCelver Discord Server or a Slack Workspace... It really comes in handy^^

captFuture, 2020-11-21:
btw... changed the lib_deps in my local repo to the latest pio requirements as you have in your repo

lib_deps = greiman/SdFat@^2.0.1
	teemuatlut/TMC2130Stepper@^2.5.1
	teemuatlut/TMCStepper@^0.7.1
	olikraus/U8g2@^2.28.8
	olikraus/U8glib@^1.19.1
	bitbucket-fmalpartida/LiquidCrystal@^1.5.0

cad435, 2020-11-21:
you can just ommit the @^Version and I think it would always use the latest version of that library

i-make-robots, 2020-11-24:
I'm out of date on this conversation.  Like this server? https://discord.gg/WrAqJwDG

i-make-robots, 2020-11-24:
I just installed PlatformIO and VSCode.  Going to try to make the leap in the morning. (11pm here)

captFuture, 2020-11-24:
Perfect, I am on discord now and this is often better for a quick chat instead of writing here :D
For starting with Platformio you can either use the tree of @cad435 https://github.com/cad435/Makelangelo-firmware/tree/PlatformIO which currently does not have your latest fixes or go with my current approach and use your repo.
https://github.com/captFuture/Makelangelo-firmware-pio (the readme contains setup info)

cad435, 2020-11-24:
Im on the discord too and just updating my PIO-Branch to the latest updates...

cad435, 2020-11-24:
For a test with my branch (I'll describe is with that one as this is simpler to do imo, wheres @captFuture Approach is trickier, but the "cleaner" way) you have to 
1) Download zip and unzip it or clone it from github
2) open VisualStudioCode + PlatformIO
3) in PIO, Klick "OpenProject"
4) Navigate to the root folder of the unzipped code (eg. where the platformio.ini file is located)
5) have fun :)

If your familiar with PIO and want to stay there, I can make a PR wich will have folder restructures for better overview etc.

i-make-robots, 2020-12-23:
Please use the platformIO branch to make any required changes.
Please write a "getting started" wiki page for dummies like me that are still trying to get running.

i-make-robots, 2021-08-14:
This is done!  Thanks, everyone.

closed 2021-08-14 by i-make-robots