MarginallyClever/Makelangelo-firmware #98: Drawing from SD card gets stuck after first line

captFuture, 2020-11-23:
After many fixes last "night" from you Dan most of the things are pretty good.
I recognized that my generated *.ngc files from makelangelo-software do not work any more.

Steps to reproduce the behavior:
1. Home Makelangelo
2. Select file from SDcard (i use my standard demo file [http://www.tarantl.com/mandala.ngc](http://www.tarantl.com/mandala.ngc) )
3. confirm start
4. Makelangelo moves to starting point, lowers pen, draws one line segment and silently stands still

When I use the PC to send over the code it is working as expected.

@cad435 can you reproduce this? btw. I changed the compiling on my PIO fork again and now it uses pre and post python scripts which copy the relevant configs before building and upon finish moves the compiled files to a release folder...
(seems you are not so far away - when germany is correct in your github profile)

i-make-robots, 2020-11-23:
I have seen this when using Adafruit SdFAT library.  In the serial monitor it will have a message about EOF.  You may also find it has trouble reading filenames from the card and you have unplug/replug the card to get a read.

Use the Grieman library instead and the problems go away.  A note about this has already been put in the README about dependencies.

captFuture, 2020-11-23:
Thanks Dan, but have to say that I am definitely using the Greiman one (greiman/SdFat@^2.0.1) Tried this with Arduino ide and Platformio. 
You are right, there is an EOF in the serial monitor, maybe something got broken yesterday when you did your fixes.

i-make-robots, 2020-11-23:
The sdcard.* files haven't changed in at least 7 months.  on my end the only thing I changed was the choice of SdFAT library.  I will test again later today, have to pick parts for a major order first.

cad435, 2020-11-24:
@captFuture I can't verify anything physically, I'm waiting on receiving a Sixi-2. Right now I only making "Pre-Work" for getting started on receival :)

i-make-robots, 2020-12-23:
Is this fixed and done?

captFuture, 2021-08-23:
Confirmed fixed and done in latest Master. Just tested it :D

closed 2021-08-23 by i-make-robots