MarginallyClever/Makelangelo-firmware #111: new-servo-system gcode M280 Snn gets no reponse.

i-make-robots, 2021-07-11:
**To Reproduce**
1. load firmware onto makelangelo robot (or mega2560+ramps+2 steppers+9g servo).  please refer to https://mcr.dozuki.com/c/Makelangelo_3_and_5_Guide for steps.
2. open serial connection to robot.
3. send M280 S90.  servo should move to 90 degrees absolute.  for me, nothing happens.
4. send M280 S50.  servo should move to 50 degrees absolute.  for me, nothing happens.

**Platform (please complete the following information):**
 - Device: makelangelo 5 or equivalent
 - OS: Windows 10
 - IDE: VSCode + platformio

**Additional context**
M280 code is in parser.cpp:1021

```void Parser::M280() {
#if NUM_SERVOS > 0
  SERIAL_ECHOPGM("M280");
  
  int id = parseInt('P', 0);
  int v = parseInt('S', servo[0].read());
  if(id <0 || id>=NUM_SERVOS) return;
  if(v>=0 && v <=180) {
    MOVE_SERVO(id,v);
  } else if(v>500) {
    servo[id].writeMicroseconds(v);
  }

  SERIAL_ECHOPAIR(" P",id);
  SERIAL_ECHOLNPAIR(" S",v);
#endif
}```

captFuture, 2021-08-23:
Manual Commands M280 S90 and M280 S50 work for me on latest Master.
In Makelangelo-Software 7.24.2 Moving servo by button is not working -> a G0 command is sent G0 F800 Z30 ...
When in Settings of Makelangelo Software the TEST buttons are not sending any command.

There probably is a misalignment of commands between the different versions of Firmware and Software -> on latest *.jar all button commands are doing nothing any more (took your compiled https://www.dropbox.com/s/8c28kdqumvlfuy5/Makelangelo-7.25.0-with-dependencies.jar?dl=0)

i-make-robots, 2021-08-23:
Thanks, @captFuture .  The dropbox file runs java 15 version, which is being totally stupid about service loaders and modules.  I have since worked it to support both.  Java is usually pretty smart, but this is not.

It knows about the modules.  it provides them.  it uses them.  and it still needs to be told there's a second list of services available.  Okay.