MarginallyClever/Makelangelo-firmware #113: All branches: High speed movement jerky despite plan-ahead

i-make-robots, 2021-07-12:
**Describe the bug**
Code that should run smoothly on this hardware - tested to run smoothly with Marlin installed - is instead jerky and erratic in Makelangelo-firmware.

**To Reproduce**
1. load firmware onto makelangelo robot (or mega2560+ramps+2 steppers+9g servo). please refer to https://mcr.dozuki.com/c/Makelangelo_3_and_5_Guide for steps.
2. Run Makelangelo-software.
3. Connect to robot.
4. open the log window to watch reports from the robot.  pay attention to OK Pnn Bnn.  P is the space in the movement buffer (low is good).  B is the space in the gcode buffer (low is good).
4. Load "thank-you.svg" and start drawing.
5. click LCD dial when prompted and watch results

areas with many short gcode commands produce jerky results even though the P and B values are very low.  well planned movement in a nearly straight line should be fluid and not-jerky.

**Platform (please complete the following information):**
 - Device: makelangelo 5 or equivalent
- OS: Windows 10
- IDE: VSCode + platformio

**Additional context**
The stepper motors are driven by an ISR on Timer1.
I thought it might be timers interfering with each other, so first I replaced the Serial handler (timer0) with Marlin's version.  No difference.
I tried merging in Marlin's servo handler on Timer3.  So far the servo does not work as expected, see issue #111 and #112.

i-make-robots, 2021-07-12:
Slightly improved by fix to #112 but not done yet.

i-make-robots, 2021-08-13:
@coliss86 if you have any insight here, that would be incredible.  I've looked high and low and I can't see what is wrong.

coliss86, 2021-08-14:
Unfortunatelly, I'm not good at C programming, I only do copy/paste 🤣 with stack overflow.

Nevertheless, I'm used to 3D printing and your problem let me think of a bandwith issue. It is common issue with octoprint on a raspberry pi with a 3D printer : the serial communication between the 2 is not fast enough and it produces jerky movement on small shape: too many instructions have to be sent to the microcontroller and they are executed fast enough thus the buffer is empty, so the robot is waiting for the next instruction. Is your communication speed the same with marlin and with Makelangelo-firmware ?

Another ideas : 
- did you try to disable the LCD ? I think it consume a lot of interrupts and maybe it "disturbs" the timing for servo.
- I do not know if it possible, but did you try to upload the gcode file on the SD card and tells the firmware to read from the SD card ? (linked with bandwith issue exposed above)

i-make-robots, 2021-08-14:
As card yes, problem persists.
Lcd panel no, I never considered it. Will do.

i-make-robots, 2021-08-15:
In my tests the segment buffer and the gcode buffer are nearly full when the jerkiness happens.   so it's not serial bandwidth that's the issue.  I have more machines to assemble this week, i'll try a no-LCD build at that time.

coliss86, 2021-08-16:
Another idea : have you tried to disable debug output sent over serial ? It may consume to much CPU...

i-make-robots, 2021-08-16:
I added the debug after the problem to see what's going on.  so the problem
existed before and without serial.