MarginallyClever/Makelangelo-firmware #120: Compilation failure

coliss86, 2021-08-13:
**Describe the bug**
I cannot compile the project on the master branch. [The CI too](https://github.com/MarginallyClever/Makelangelo-firmware/runs/3246095083?check_suite_focus=true).

**To Reproduce**
Run `pio run` in the folder of the firmware and I get the following error : 
```
pio/libdeps/rumba-sixi3/TMCStepper/src/TMCStepper.h:25:11: fatal error: SoftwareSerial.h: No such file or directory

************************************************************************
* Looking for SoftwareSerial.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:SoftwareSerial.h"
* Web  > https://platformio.org/lib/search?query=header:SoftwareSerial.h
*
************************************************************************

  #include <SoftwareSerial.h>
           ^~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio/build/rumba-sixi3/libda9/TMCStepper/source/ENCMODE.cpp.o] Error 1
=============================================== [FAILED] Took 2.08 seconds ===============================================

Environment    Status    Duration
-------------  --------  ------------
rumba-sixi3    FAILED    00:00:02.076
========================================== 1 failed, 0 succeeded in 00:00:02.076 ==========================================
```

**Expected behavior**
The project compiles fine

**Additional context**
I made a first PR to build all environnement https://github.com/MarginallyClever/Makelangelo-firmware/pull/119

i-make-robots, 2021-08-13:
should be fixed by df378efe8a87b95f6d6c2d7d711c0dc121b7e873.  can you confirm?

i-make-robots, 2021-08-13:
I commented out inclusion of TMCStepper libraries after seeing that Marlin has the same issue, as of yet unresolved. https://github.com/teemuatlut/TMCStepper/issues/56

coliss86, 2021-08-14:
Thanks !
It is getting better but [not all are OK](https://github.com/MarginallyClever/Makelangelo-firmware/runs/3326531149).

In https://github.com/MarginallyClever/Makelangelo-firmware/pull/119 I have enabled the compilation of all supported board, you can see what is wrong here: on the home page of the project, you will see a red cross : 
[image]
then click on it : 
[image]

Then you can click on the "details" on the line with red cross : 
[image]

You can also click on the "action" tab directly on the home page.

You'll have the log of what has been compile for each commit.

To sum up, the following boards still do not compile:
- BIGTREE_SKR_PRO-Sixi3
- BIGTREE_SKR_PRO-Stewart
- sixi2

i-make-robots, 2021-08-14:
Yeah… every run sends me an email with results. Super fun. I’ll look at it in an hour.

i-make-robots, 2021-08-14:
Ugh... making it work for STM is so.  much.  work.

i-make-robots, 2021-08-15:
The HAL is basically copied from Marlin.  What M-F does differently is implement up to 6 axies and no extruder.

i-make-robots, 2021-08-23:
STM is still open.  I guess that's my work for the week.