MarginallyClever/Makelangelo-firmware #122: Help for compilation for cnc shield

coliss86, 2021-08-21:
**Describe the bug**
As stated in the readme, the CNC Shield v3 is supported but I cannot compile the firmaware for it.

**To Reproduce**

In the `platform.io`, I have added a new `env` called `cncv3-polargraph` and configured as followed:
```
[env:cncv3-polargraph]
  platform = atmelavr
  board = nanoatmega328
  extends = common_avr8
  build_flags =
  	${common.build_flags}
        -DHEADER_RAMPS
  	-DMACHINE_STYLE=POLARGRAPH
  	-DMOTHERBOARD=BOARD_CNCV3
        -DMACHINE_HARDWARE_VERSION=CUSTOM
  	-DTARGET_DEFAULT
        -DUSE_ALT_SERVO
  lib_deps =
```
`MACHINE_HARDWARE_VERSION` is set to `CUSTOM` in order to skip the definition of the sdcard and lcd in `src/robots/polargraph.h`.

I do not know if want I'm doing is correct or not.

The compilation failed ( [full report](https://github.com/coliss86/Makelangelo-firmware/runs/3390036147) ) with:
```
Run pio run --environment ${MATRIX_ENV}
Processing cncv3-polargraph (platform: atmelavr; board: nanoatmega328; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/nanoatmega328.html
PLATFORM: Atmel AVR (3.3.0) > Arduino Nano ATmega328
HARDWARE: ATMEGA328P 16MHz, 2KB RAM, 30KB Flash
DEBUG: Current (avr-stub) On-board (avr-stub, simavr)
PACKAGES: 
 - framework-arduino-avr 5.1.0 
 - toolchain-atmelavr 1.70300.191015 (7.3.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 5 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SPI> 1.0
|-- <EEPROM> 2.0
|-- <Wire> 1.0
Building in release mode
Compiling .pio/build/cncv3-polargraph/src/Makelangelo-firmware.cpp.o
Compiling .pio/build/cncv3-polargraph/src/boards/HAL_avr.cpp.o
Compiling .pio/build/cncv3-polargraph/src/boards/HAL_stm32f4.cpp.o
Compiling .pio/build/cncv3-polargraph/src/boards/MarlinSerial.cpp.o
Compiling .pio/build/cncv3-polargraph/src/boards/skrpro.cpp.o
In file included from /home/runner/.platformio/packages/framework-arduino-avr/cores/arduino/Arduino.h:30:0,
                 from src/boards/HAL_avr.h:5,
                 from src/boards/cncv3.h:57,
                 from src/configure.h:100,
                 from src/boards/MarlinSerial.cpp:39:
src/boards/MarlinSerial.cpp: In function 'void USART0_RX_vect()':
src/boards/MarlinSerial.cpp:558:20: warning: 'USART0_RX_vect' appears to be a misspelled 'signal' handler, missing '__vector' prefix [-Wmisspelled-isr]
 ISR(SERIAL_REGNAME(USART, SERIAL_PORT, _RX_vect)) {
                    ^
src/boards/MarlinSerial.h:69:54: note: in expansion of macro '_SERIAL_REGNAME'
   #define SERIAL_REGNAME(registerbase,number,suffix) _SERIAL_REGNAME(registerbase,number,suffix)
                                                      ^~~~~~~~~~~~~~~
src/boards/MarlinSerial.cpp:558:5: note: in expansion of macro 'SERIAL_REGNAME'
 ISR(SERIAL_REGNAME(USART, SERIAL_PORT, _RX_vect)) {
     ^~~~~~~~~~~~~~
```

All the modifications are in https://github.com/MarginallyClever/Makelangelo-firmware/pull/121

**Expected behavior**
The project compiles fine

**Additional context**
I bought this kit: https://www.aliexpress.com/item/1005002826643730.html
[image]

The pinout seems identical to the cnc shield v3, it may work :)

wenoptics, 2021-08-22:
I am also looking into `Mega2560 + CNC Shield v3` support, which looks like not included in the release.

i-make-robots, 2021-08-22:
The cnc shield is the speed limit for drawing. It might be done but requires careful tuning.

i-make-robots, 2021-08-22:
Ok, I got confused and somehow thought the CNC shield was the AMS shield.  my bad.

i-make-robots, 2021-08-22:
In 83d05b0 I added cnc3-polargraph to platform.io and changed the requirements such that it MUST use a mega2560 (the minimum).

Now I need some help from you.  The board* files are included before the robot* files.  what that means is board_cnc3 is included first, then robot/polargraph.h.  but polargraph.y is where it it says "oh, you want makelangelo 5?  Well that has an SD card and an LCD."  which it does NOT if you are building on the cnc3 platform.  SO the question is how do we rearrange these includes to work nicely for everyone?

i-make-robots, 2021-08-22:
moved HAS_SD and LCD_TYPE to platform.io so that it should work with cnc3-polargraph.  Please run it and correct pin settings, etc.

captFuture, 2021-08-23:
I just tried to compile for an Arduino Nano as requested here if I interpret the photos of the board.
The issue obviously is the Serial Interrupt in the MarlinSerial library which is not working.
The sketch itself fits if the alternate Servo is disabled.
I am not sure if it is possible to rewrite the MarlinSerial for an Atmega328p because marlin is only running on Arduino Mega boards.

i-make-robots, 2021-08-23:
328p iirc doesn’t have enough clocks. Marlin. Cannot run on 328p either.

i-make-robots, 2021-08-23:
Ugh... my brain isn't what it used to be.  Even with your pictures my brain still said "cnc shield?  You mean this?" 

https://hacksterio.s3.amazonaws.com/uploads/attachments/1126097/989baf7d6a1b68b8c5ddae17796a1587_FcC732JswP.png

which can be run on a mega.

i-make-robots, 2021-08-23:
the challenge with this shield, for me, was hooking up the mega and the shield to power, as they do not share.  no way am I putting two power supplies on this thing.

coliss86, 2021-08-23:
> moved HAS_SD and LCD_TYPE to platform.io so that it should work with cnc3-polargraph. Please run it and correct pin settings, etc.

I think LCD_TYPE has been missing in https://github.com/MarginallyClever/Makelangelo-firmware/commit/fdaade1867fd7d776d637b81b03e26aa604d7f97. Only HAS_SD has been moved to platformio.ini

coliss86, 2021-08-23:
> Ugh... my brain isn't what it used to be. Even with your pictures my brain still said "cnc shield? You mean this?"
> 
> [image]
> 
> which can be run on a mega.

Thanks for your clarification, I though the shield + nano was enough for this firmware, but you say you say it doesn't. My bad :'(

coliss86, 2021-08-23:
I close the issue as you help me a lot :)

closed 2021-08-23 by coliss86